kevinzakka/mink #7: [New Example]: xArm7 Lift Task, Gripper doesn't work

jonzamora, 2024-07-26:
Hi @kevinzakka 

I'm working on an example where an `xArm7` can grasp and lift a cube with the help of `mink`, however, I ran into an issue with the gripper.

https://github.com/user-attachments/assets/a3200082-b9ba-478d-b977-4c763f023c6f

**Main Question**: Could you help me debug the `xArm7` gripper so it can grasp the cube correctly? 

**Things i've tried**: I tried all the suggestions mentioned in https://github.com/google-deepmind/mujoco/issues/786 and https://github.com/google-deepmind/mujoco_menagerie/issues/54, though my attempts were unsuccessful so far.

**Link to my code w/ Minimal Example**: My code can be found here https://github.com/jonzamora/mink.

To reproduce the video above, simply run `mjpython examples/arm_xarm7.py`.

kevinzakka, 2024-07-28:
Hi @jonzamora,

It's great to see you've already started using mink for your own project! This issue is more of a MuJoCo/Menagerie modeling problem than a mink one. That being said, I think the most likely culprit is that the geoms of the fingertips of the xarm7 need to be improved. If you raise an issue in Menagerie, I can properly take a look and respond. Sound good?

jonzamora, 2024-07-28:
Sounds great @kevinzakka ! I'll raise an issue in the Menagerie and close the issue here : )

closed 2024-07-28 by jonzamora