kevinzakka/mink #13: Creating IK function for Biarm

Kelym, 2024-08-05:
Hi Kevin,
Thanks for sharing your work. I am trying to create an IK function for aloha. I followed your example script and successfully created an IK solver for UR5. But when I applied the same logic to create an IK function to aloha - I got some error in the solver:
```
        tasks = [
            left_end_effector_task := mink.FrameTask(
                frame_name="left/attachment_site",
                frame_type="site",
                position_cost=1.0,
                orientation_cost=1.0,
                lm_damping=1.0,
            ),
        ]

        # IK settings.
        solver = "osqp"

        # Update initial joint positions
        if current_joint_positions is not None:
            self.configuration.update(current_joint_positions)

        T_wt_left = self.create_se3_from_ee_pose(ee_poses[0:7])
        left_end_effector_task.set_target(T_wt_left)
rate_dt = 0.01

        # Compute velocity and integrate into the next configuration.
        for i in range(max_iters):
   -->      vel = mink.solve_ik(
                self.configuration, tasks, rate_dt, solver, damping=1e-3, limits=self.limits
            )

File ~/miniforge3/envs/monopi/lib/python3.11/site-packages/mink/solve_ik.py:104, in solve_ik(configuration, tasks, dt, solver, damping, safety_break, limits, **kwargs)
    102 dq = result.x
    103 assert dq is not None
--> 104 v: np.ndarray = dq / dt
    105 return v

TypeError: unsupported operand type(s) for /: 'NoneType' and 'float'"
```

The target ee pose I sent was taken from the robot's current EE pose and I believe it is solvable. I also tried the `osqp` solver as in your dual iiwa example. Is there anything I am missing?

Here's the UR5 IK/FK function I created for reference. 

```
class MinkRobot:
    def __init__(self, xml_path):
        self.model = mujoco.MjModel.from_xml_path(xml_path)
        self.configuration = mink.Configuration(self.model)
        self.data = self.configuration.data
        self.set_up_mink()

    def set_up_mink(self):
        collision_pairs = [
            (["wrist_3_link"], ["floor"]),
        ]

        limits = [
            mink.ConfigurationLimit(model=self.model),
            mink.CollisionAvoidanceLimit(model=self.model, geom_pairs=collision_pairs),
        ]

        max_velocities = {
            "shoulder_pan_joint": np.pi,
            "shoulder_lift_joint": np.pi,
            "elbow_joint": np.pi,
            "wrist_1_joint": np.pi,
            "wrist_2_joint": np.pi,
            "wrist_3_joint": np.pi,
        }
        velocity_limit = mink.VelocityLimit(self.model, max_velocities)
        limits.append(velocity_limit)

        self.limits = limits

    def mink_fk(self, joint_angles):
        fk_pose = np.zeros(7)
        self.configuration.update(joint_angles)
        fk_pose_se3 = self.configuration.get_transform_frame_to_world("attachment_site", "site").wxyz_xyz

        fk_pose[:3] = fk_pose_se3[4:7]
        fk_pose[3:6] = fk_pose_se3[1:4]
        fk_pose[6] = fk_pose_se3[0]
        return fk_pose

    def mink_ik(self, target_ee_pose, current_joint_positions=None, pos_threshold=1e-4, ori_threshold=1e-4, max_iters=20):
        tasks = [
            end_effector_task := mink.FrameTask(
                frame_name="attachment_site",
                frame_type="site",
                position_cost=1.0,
                orientation_cost=1.0,
                lm_damping=1.0,
        )]

        # IK settings.
        solver = "quadprog"

        # Update initial joint positions
        if current_joint_positions is not None:
            self.configuration.update(current_joint_positions)

        target_ee_pose_matrix = np.zeros((4, 4))
        target_ee_pose_matrix[:3, :3] = Rotation.from_quat(target_ee_pose[3:]).as_matrix()
        target_ee_pose_matrix[:3, 3] = target_ee_pose[:3]
        T_wt = mink.SE3.from_matrix(target_ee_pose_matrix)
        end_effector_task.set_target(T_wt)

        rate_dt = 0.005

        # Compute velocity and integrate into the next configuration.
        for i in range(max_iters):
            vel = mink.solve_ik(
                self.configuration, tasks, rate_dt, solver, damping=1e-3, limits=self.limits
            )
            self.configuration.integrate_inplace(vel, rate_dt)
            err = end_effector_task.compute_error(self.configuration)
            pos_achieved = np.linalg.norm(err[:3]) <= pos_threshold
            ori_achieved = np.linalg.norm(err[3:]) <= ori_threshold
            if pos_achieved and ori_achieved:
                return self.data.qpos

        fk_pose_se3 = self.configuration.get_transform_frame_to_world("attachment_site", "site").wxyz_xyz
        print(f'target_ee_pose {target_ee_pose}')
        print(f'Achieved: xyz {fk_pose_se3[4:7]}, qxqyqz {fk_pose_se3[1:4]}, qw {fk_pose_se3[0]}')
        print(f'{self.data.qpos}')
        return None
```

kevinzakka, 2024-08-05:
Hi Kay, it's great to hear from you again and glad to hear you're using mink :)

I just create a PR (#14) that demonstrates how to setup mink for ALOHA. I thought it would be easier for you to compare with something working to quickly spot any error. Let me know if you have any further questions!

kevinzakka, 2024-08-05:
Oh and I added collision avoidance as an added bonus!

kevinzakka, 2024-08-06:
Marking as closed by #14.

closed 2024-08-06 by kevinzakka

Kelym, 2024-08-06:
Awesome thanks so much!!
BTW do you have some clue on what caused the error I ran into?

kevinzakka, 2024-08-06:
@Kelym I think I need more context but from what you showed me in the code snippet above, `frame_name="left/attachment_site"` seems slightly suspicious. Did you create such a site in the ALOHA xml?

kevinzakka, 2024-08-06:
@Kelym Another thing is that the quaternion convention in MuJoCo / mink is `wxyz` but it looks like scipy's `from_quat` uses the [default `scalar_first=False`](https://docs.scipy.org/doc/scipy/reference/generated/scipy.spatial.transform.Rotation.from_quat.html). That might also contribute to an invalid target.

Kelym, 2024-08-07:
Thanks for the suggestion!

Re: attachment. Yes I created the attachment as a site in the ALOHA xml - I see that you have been using mocap (left/gripper) to set the target, is there a reason to not set the target using a site that's directly attached to the robot body?
Re: angle convention. I noticed that mink returns wxyz_xyz so I have been processing them back to xyz_xyzw & vice versa. My UR5 code seemed to work fine after the convention. 

I really appreciate you creating and sharing the Aloha example. Just doing this to understand what might have went wrong.