kevinzakka/mink #23: Is it possible to do a conda install version

kelejichi6463, 2024-08-26:
Hello author, I get an error when installing with PIP that there is no suitable mink version, so I would like to ask you if it is possible to make an installation version of conda. Thank you very much

jonzamora, 2024-08-26:
Yeah, you can try this `environment.yaml`

```YAML
name: mujoco
channels:
  - conda-forge
  - defaults
dependencies:
  - python=3.10
  - pip
  - pip:
      - mink[examples]
      - 
```

Then, just run `conda` or [`mamba`](https://mamba.readthedocs.io/en/latest/) like so: `mamba env create -f environment.yaml` and `mamba activate mujoco`. You should be able to run `mink` and its related scripts after this.

kevinzakka, 2024-08-26:
@kelejichi6463 Can I have some more information? Python version, OS, commands run, etc.

kelejichi6463, 2024-08-27:
@jonzamora It works! Great!  Thank you jonzamora!

kelejichi6463, 2024-08-27:
@kevinzakka Hi, kevinzakka, My python version is 3.10.14, OS is ubuntu, and I make a env for pink, it works well~

kelejichi6463, 2024-08-27:
@jonzamora  This method can be successfully installed on one computer, but it requires additional manual installation of some packages. When installing on another computer, an error occurred midway, and reinstallation prompted: CondaValueError: prefix already exists: /home/XXX/anaconda3/envs/mink .

jonzamora, 2024-08-27:
@kelejichi6463 that must mean you have an existing `anaconda` package manager installed on that machine. If you want to have a clean install with `mamba`, you can first remove `anaconda` by following their instructions [here](https://docs.anaconda.com/anaconda/install/uninstall/).

The key commands would be

```bash
conda activate
conda init --reverse --all
rm -rf /home/XXX/anaconda3
```

This might not cover it all, and it's out-of-scope probably for what is related to `mink`, but hopefully this helps. Cheers!

jonzamora, 2024-08-27:
@kelejichi6463 Try to update the environment like this

```bash
mamba env update -f environment.yaml
# or
conda env update -f environment.yaml
```

Assuming you just need to update the existing `mink` environment.

kevinzakka, 2024-09-09:
I can't really reproduce this issue. mink has worked for me on multiple platforms with the only requirement being python>=3.9. Going to close, but feel free to re-open if there are other issues that persist with a clean environment install.

Also thanks @jonzamora for the help!

closed 2024-09-09 by kevinzakka