kevinzakka/mink #35: interactive viewer to control the arm motion

vidhiJain, 2024-10-25:
I am curious if it is possible to have interactive mujoco viewer to move the end effector and have the IK solved for it.

kevinzakka, 2024-10-25:
Hi @vidhiJain, the examples in the examples dir all use the interactive viewer with mocap bodies that you can move around to specify targets. Try them
and let me know if that solves your problem.

vidhiJain, 2024-10-25:
I'm running on macOS `mjpython arm_aloha.py`, `mjpython arm_ur5e.py`. It displays the end-effector highlighted but it seems I can't control it (with laptop trackpad). 
For `mjpython mobile_stretch.py`, the window opens with the robot rotating to reach hand and base end-effectors.
Can we move the red box interactively on the viewer? If yes, how? 
I suspected it won't because the viewer is launched as `passive`.

kevinzakka, 2024-10-25:
@vidhiJain Double click the mocap body to highlight it. Then hit ctrl + left to rotate and ctrl + right to translate.

jonzamora, 2024-10-25:
^ This is also in the MuJoCo documentation, see [here](https://mujoco.readthedocs.io/en/stable/programming/samples.html#sasimulate). 

> "Interaction is done with the mouse; built-in help with a summary of available commands is available by pressing the `F1` key. Briefly, an object is selected by left-double-click. The user can then apply forces and torques on the selected object by holding Ctrl and dragging the mouse. Dragging the mouse alone (without Ctrl) moves the camera."

vidhiJain, 2024-10-25:
Thank you for the prompt response~!

closed 2024-10-25 by vidhiJain

kevinzakka, 2024-10-25:
Thank you @jonzamora !!