kevinzakka/mink #36: Attaching the gripper without extending qpos?

benthebear93, 2024-10-27:
Hi, I am a PhD student trying to use MuJoCo and mink for maniplulation simulation.
Thank you for your work :)

I have been trying to attach the robotiq_2f85 gripper from the [mujoco_menagerie](https://github.com/google-deepmind/mujoco_menagerie/tree/main/robotiq_2f85) to ur5e from your example code.

I was somewhat able to attach the gripper(using weld) but i am getting "Invalid qpos size error" which makes seens as i added the gripper. 

`ValueError: Error: keyframe 0: invalid qpos size, expected length 21
Element name 'home', id 0
`
Althought it might not be the issue of mink but is there any clever ways i can use mink only for the manipulator?

kevinzakka, 2024-10-27:
Hi @benthebear93, I wouldn't recommend attaching the gripper with a weld constraint. You have a few options:

1. Use the newly introduced [MjSpec](https://mujoco.readthedocs.io/en/latest/programming/modeledit.html) mechanism to attach the gripper to the arm.
2. Use [PyMJCF](https://github.com/google-deepmind/dm_control/tree/main/dm_control/mjcf#pymjcf) (dm_control module) to attach it. Read this [FAQ](https://github.com/google-deepmind/mujoco_menagerie/blob/main/FAQ.md#how-do-i-attach-a-hand-to-an-arm) for an explanation of how to do it. On top of that, a few other examples in `mink` leverage this functionality, see for example the [iiwa + allegro hand example](https://github.com/kevinzakka/mink/blob/main/examples/arm_hand_iiwa_allegro.py#L29).

Regardless of the method you choose, dealing with keyframes is a bit finnicky at the moment so I would recommend removing it from the model ([here](https://github.com/kevinzakka/mink/blob/main/examples/universal_robots_ur5e/ur5e.xml#L131)) and just define and use it in python.

benthebear93, 2024-10-27:
Thank you for the good advice. I always kind of feel lost when I use MuJoCo because there are so many ways to approach it. I’ll take a look at the links you gave me. Thank you! :) 👍

benthebear93, 2024-10-28:
Hi @kevinzakka, I've successfully attached the robotiq_2f85 to the UR5e. I have another question about the VelocityLimit function. Although the robotiq_2f85 has 8 joints, I mostly use a single actuator to control the fingers. However, the VelocityLimit function only checks the joint limits, which results in a warning due to the 8 joints of the 2f85. 

While it makes sense for the function to check the gripper's joint limits since it is a joint, it feels somewhat unnecessary. **Is there anyway to make it ignore the gripper's joints?**  solve_ik provides a velocity result with 14 elements, which seems excessive for my case.

kevinzakka, 2024-10-28:
Hi @benthebear93, a bit confused about your question. The velocity limit takes in a dictionary, which allows you to pick and choose which joints you want to constrain. You can see an example of this in [arm_ur5e_actuators.py](https://github.com/kevinzakka/mink/blob/main/examples/arm_ur5e_actuators.py#L48-L57):

```python
max_velocities = {
    "shoulder_pan": np.pi,
    "shoulder_lift": np.pi,
    "elbow": np.pi,
    "wrist_1": np.pi,
    "wrist_2": np.pi,
    "wrist_3": np.pi,
}
velocity_limit = mink.VelocityLimit(model, max_velocities)
limits.append(velocity_limit)
```

benthebear93, 2024-10-29:
@kevinzakka Sorry, my mistake. It was actually the check_limits function in the configuration that triggers the WARNING because configuration.qval exceeded the joint limits. **Is it normal to see this WARNING when running solve_ik?**
`WARNING:root:Value 0.00003153 at index 11 is outside of its limits: [-1.57, 0.00]`

This doesn’t seem to happen for the UR5e joints, but it **frequently occurs with the gripper’s joints**. I could adjust the tolerance values to ignore the WARNINGs, but that feels a bit odd.

benthebear93, 2024-10-30:
https://github.com/google-deepmind/mujoco/discussions/2193

I received an answer to a similar question I asked, so I will close the issue as resolved.