kevinzakka/mink #37: Questions about Dataset Retargeting

zengweishuai, 2024-11-05:
Hi @kevinzakka , thanks you for your great work! I firmly believe that it will promote the development of the whole robot community!

There is a question I'd love to inquire. What if I want to use mink to retarget some datasets like AMASS. Is there any way to handle SMPL format data or do some transformation to let them fit mink?

I'm sorry for my disturbance as I'm new to robotics. Expect your reply!

IlanElfen, 2024-12-03:
Have you found a way to do this?

kevinzakka, 2025-02-14:
Hi @zengweishuai, I'm going to release something soon that uses mink to perform re-targeting. Will ping here when this is the case.

shaoxiang, 2025-06-22:
I am very much looking forward to the work of retargeting motion capture mocap data to humanoid robots.

kevinzakka, 2025-12-19:
I haven't had time to properly do this, but there is [GMR](https://github.com/YanjieZe/GMR) you can take a look at in the meantime.

closed 2025-12-19 by kevinzakka