kevinzakka/mink #39: Problem with rendering using a fixed camera

ErinZhang1998, 2024-11-17:
Hi,

I am wondering how I can render using a camera with property `cam_rs.type = mujoco.mjtCamera.mjCAMERA_FIXED` where `cam_rs` is a `mujoco.MjvCamera()` object.
The camera doesn't seem to update unless I run `mujoco.mj_step(self.model, self.data)`, where the `model` and `data` are from the mink configuration. However, `mujoco.mj_step` seem to mess with something that hurts the IK accuracy. The absolute error between the desired EE position and actual EE position accumulates as the trajectory gets longer.

Thank you so much!

kevinzakka, 2025-02-14:
Hi @ErinZhang1998, really sorry for the delayed response. In principle, `integrate_inplace` should be sufficient since it calls `mj_kinematics` under the hood. Can you provide some more information and ideally a minimum working example I can debug?

kevinzakka, 2025-05-23:
Going to close due to inactivity.

closed 2025-05-23 by kevinzakka