kevinzakka/mink #41: Failed to get the frame of the mocap object from the configuration

gaotian9919, 2024-12-04:
Hi, I'm trying to control my dexterous hand model with mink, I'm only using mujoco's python bindings, as arm_ur5e_actuators.py in examples.
I've found that when I use print(configuration.get_transform_frame_to_world("target","body")) in the loop statement of the arm_ur5e_actuators.py to get the frame of the mocap object, the position result is fixed and doesn't change as I move the target box, while another example, arm_ hand_iiwa_allegro.py, however, I can get the position of the mocap object that updates with motion.
I found a difference between where they build model and data, the latter using the mjcf package in the dm_control, but I didn't have a deeper understanding of the problem.
Pls is it normal to have the issue of configuration.get_transform_frame_to_world not updating with just mujoco python bindings, which prevents me from mounting my custom dexterous hand model on the robot model
My native language is not English, please forgive me if there is a translation error, thank you for your work!

gaotian9919, 2024-12-04:
I find the answer，I need to replace the model and data of the Viewer() and step() with configuration.model and configuration.data so that the obtained frame will be updated.
Could you explain the purpose of using dm_control.mjcf in arm_hand_iiwa_allegro.py examples?

kevinzakka, 2025-02-14:
Hi @gaotian9919, I used `mjcf` to programmatically attach the hand to the arm. At the time I made this, MuJoCo's new model editing API wasn't fully ready but I recommend using it now. You can read more about it [here](https://mujoco.readthedocs.io/en/stable/python.html#model-editing) and use it instead of MJCF.

closed 2025-02-14 by kevinzakka