kevinzakka/mink #42: why is the mocap position  not fixed

yijing2050, 2025-01-08:
Hello ,when I test the examples/mobile_tidybot.py, run it,the output is as follows。
pinch_site pose: [0.57666519 0.0013501  0.76872431]
the code of mobile_tidybot.py is as follows:
from dataclasses import dataclass
from pathlib import Path

import mujoco
import mujoco.viewer
import numpy as np
from dm_control.viewer import user_input
from loop_rate_limiters import RateLimiter

import mink

_HERE = Path(__file__).parent
_XML = _HERE / "stanford_tidybot" / "scene.xml"

@dataclass
class KeyCallback:
    fix_base: bool = True
    pause: bool = True

    def __call__(self, key: int) -> None:
        if key == user_input.KEY_ENTER:
            self.fix_base = not self.fix_base
        elif key == user_input.KEY_SPACE:
            self.pause = not self.pause

if __name__ == "__main__":
    model = mujoco.MjModel.from_xml_path(_XML.as_posix())
    data = mujoco.MjData(model)

    # Joints we wish to control.
    # fmt: off
    joint_names = [
        # Base joints.
        "joint_x", "joint_y", "joint_th",
        # Arm joints.
        "joint_1", "joint_2", "joint_3", "joint_4", "joint_5", "joint_6", "joint_7",
    ]
    # fmt: on
    dof_ids = np.array([model.joint(name).id for name in joint_names])
    actuator_ids = np.array([model.actuator(name).id for name in joint_names])

    configuration = mink.Configuration(model)

    end_effector_task = mink.FrameTask(
        frame_name="pinch_site",
        frame_type="site",
        position_cost=1.0,
        orientation_cost=1.0,
        lm_damping=1.0,
    )

    posture_cost = np.zeros((model.nv,))
    posture_cost[3:] = 1e-3
    posture_task = mink.PostureTask(model, cost=posture_cost)

    immobile_base_cost = np.zeros((model.nv,))
    immobile_base_cost[:3] = 100
    damping_task = mink.DampingTask(model, immobile_base_cost)

    tasks = [
        end_effector_task,
        posture_task,
    ]

    limits = [
        mink.ConfigurationLimit(model),
    ]

    # IK settings.
    solver = "quadprog"
    pos_threshold = 1e-4
    ori_threshold = 1e-4
    max_iters = 20

    key_callback = KeyCallback()

    with mujoco.viewer.launch_passive(
        model=model,
        data=data,
        show_left_ui=True,
        show_right_ui=True,
        key_callback=key_callback,
    ) as viewer:
        mujoco.mjv_defaultFreeCamera(model, viewer.cam)

        mujoco.mj_resetDataKeyframe(model, data, model.key("home").id)
        configuration.update(data.qpos)
        posture_task.set_target_from_configuration(configuration)
        mujoco.mj_forward(model, data)

        # Initialize the mocap target at the end-effector site.
        mink.move_mocap_to_frame(model, data, "pinch_site_target", "pinch_site", "site")

        rate = RateLimiter(frequency=200.0, warn=False)
        dt = rate.period
        t = 0.0
        while viewer.is_running():
            # Update task target.
            T_wt = mink.SE3.from_mocap_name(model, data, "pinch_site_target")
            end_effector_task.set_target(T_wt)

            # Compute velocity and integrate into the next configuration.
            for i in range(max_iters):
                if key_callback.fix_base:
                    vel = mink.solve_ik(
                        configuration, [*tasks, damping_task], rate.dt, solver, 1e-3
                    )
                else:
                    vel = mink.solve_ik(configuration, tasks, rate.dt, solver, 1e-3)
                configuration.integrate_inplace(vel, rate.dt)

                # Exit condition.
                pos_achieved = True
                ori_achieved = True
                err = end_effector_task.compute_error(configuration)
                pos_achieved &= bool(np.linalg.norm(err[:3]) <= pos_threshold)
                ori_achieved &= bool(np.linalg.norm(err[3:]) <= ori_threshold)
                if pos_achieved and ori_achieved:
                    break

            if not key_callback.pause:
                data.ctrl[actuator_ids] = configuration.q[dof_ids]
                mujoco.mj_step(model, data)
            else:
                mujoco.mj_forward(model, data)

            # Visualize at fixed FPS.
            viewer.sync()
            rate.sleep()
            t += dt
        print("pinch_site pose:", data.sensordata[:3])
in the examples/stanford_tidybot/tidybot.xm  I added 
 <sensor>
        <framepos objtype="site" objname="pinch_site" />
	      <framelinvel objtype="site" objname="pinch_site" />
  </sensor>
I am curious the position of mocap ,we set it pos="0.5 0 .5",as follows
 <worldbody>
    <light pos="0 0 1.5" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
    <body name="pinch_site_target" pos="0.5 0 .5" quat="0 1 0 0" mocap="true">
    
      <geom type="box" size=".05 .05 .05" contype="0" conaffinity="0" rgba=".6 .3 .3 .2"/>
    </body>
  </worldbody>
</mujoco>
after running the code,why the output is " pinch_site pose: [0.57666519 0.0013501  0.76872431]",the pinch _site pose is also the mocap pose.When I modify the pose of pinch_site_target to pos="2.5 0 1.5", the output is the same after running the code.Looking forwards to your reply! Thanks

yijing2050, 2025-01-08:
in the examples/stanford_tidybot/tidybot.xm I added
...
</worldbody>

  <sensor>
        <framepos objtype="site" objname="pinch_site" />
	      <framelinvel objtype="site" objname="pinch_site" />
  </sensor>

  <contact>
    <exclude body1="base" body2="left_driver"/>
...

I am curious the position of mocap ,we set it pos="0.5 0 .5",as follows
<worldbody>
    <light pos="0 0 1.5" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
    <body name="pinch_site_target" pos="0.5 0 .5" quat="0 1 0 0" mocap="true">
    
      <geom type="box" size=".05 .05 .05" contype="0" conaffinity="0" rgba=".6 .3 .3 .2"/>
    </body>
  </worldbody>

[image]

kevinzakka, 2025-02-14:
Hi @yijing2050, could you clarify your question, I don't really understand what the problem is.

kevinzakka, 2025-05-23:
Going to close due to inactivity.

closed 2025-05-23 by kevinzakka