kevinzakka/mink #43: Closed loop not fixed using mink

JerryYan24, 2025-02-14:
Hi, i add some <equality> in my mjcf file:
`<equality>
        <connect name="kinematic_link" active="true" body1="link_3" body2="link_4" anchor="0 0 2.0" />
    </equality>` 
but when i do ik calculation using mink, those equality are not working. Is there any idea how to solve it, thanks.

kevinzakka, 2025-02-18:
Hi @JerryYan24, I believe adding a `RelativeFrameTask` that constrains the two bodies together at the connect site (with a high enough cost) should work. Can you let me know if that is the case?

simeon-ned, 2025-02-27:
@kevinzakka Great! I’ve [opened a PR](https://github.com/kevinzakka/mink/pull/49). It’s missing tests for now, and I was also thinking it might be worth restructuring a bit—for example, moving everything related to the constraint mask, etc., into the configuration to keep it more in line with other tasks. But let’s discuss this directly in the PR—I’m happy to make any changes if needed!

kevinzakka, 2025-05-23:
Fixed via #63.

closed 2025-05-23 by kevinzakka