kevinzakka/mink #45: mink with mujoco playground

gemcollector, 2025-02-24:
Hi, zakka. I wanna ask can mink be used in mujoco playground? I notice that mink is based on numpy while mujoco playground is based on jax ? Can we directly transfer mink to the mujoco playground for parallel training?  Thanks!

kevinzakka, 2025-02-24:
For direct end effector control, no. If you’re retargeting trajectories and trying to learn it deepmimic style, you can do the retargeting offline and then use that data in playground.

kassasin, 2025-02-25:
> For direct end effector control, no. If you’re retargeting trajectories and trying to learn it deepmimic style, you can do the retargeting offline and then use that data in playground.

Do you have any plan to rewrite this to jax for using GPU to accelerate?

domrachev03, 2025-02-27:
Good evening! We with my team actually worked for a while on such tool, and we have released it recently. It closely follows mink, but everything is implemented using `jax` and `mujoco mjx`. We called it [mjinx](https://github.com/based-robotics/mjinx/tree/main).

You can check the description on [the documentation page](https://based-robotics.github.io/mjinx/), and check some of the examples (which include batched [manipulator](https://github.com/based-robotics/mjinx/blob/main/examples/local_ik_vmapped_output.py), [quadruped](https://github.com/based-robotics/mjinx/blob/main/examples/go2_squat.py) and [biped](https://github.com/based-robotics/mjinx/blob/main/examples/g1_heart.py) IK). 

Hope you find it useful!

[image]

kevinzakka, 2025-02-27:
@domrachev03 This is awesome, wanna make a post in Announcements here and / or on MuJoCo?

simeon-ned, 2025-02-27:
@kevinzakka I hope @domrachev03  doesn’t mind me answering here :) We’d be very happy with that! It would be really helpful if you could mention this project both in MuJoCo and MINK. Thank you so much!

kassasin, 2025-02-27:
> Good evening! We with my team actually worked for a while on such tool, and we have released it recently. It closely follows mink, but everything is implemented using `jax` and `mujoco mjx`. We [...]

It is very useful and awesome!

domrachev03, 2025-03-01:
Hi @kevinzakka, thank you for clarification! I've created posts both [here](https://github.com/kevinzakka/mink/discussions/52) and in [MuJoCo repository](https://github.com/google-deepmind/mujoco/discussions/2466). Unfortuately, only maintainers are allowed to make announcements in repos, so I made them in general disscussion segment. It would be lovely, if any of maintainers could transfer them in the Announcement category, if they find it suitable. 

Once again, thank you for your support, `mink` highly inspired us for the development of `mjinx` and helped implementing many components using `mjx`.

closed 2025-03-15 by gemcollector