kevinzakka/mink #64: mink with another freejoint body

gemcollector, 2025-03-15:
Hi, I setup a franka arm with dexterous hand and an object with freejoint. But I found that with this freejoint object, the mink will fail to solve the right solution. But if I remove the object the simulation is correct. How should I avoid the impact from the freejoint object. 

```
<mujoco model="franka_lift_dex_scene">

  <include file="basic_scene.xml"/>
  <include file="panda_with_dex_old.xml"/>

  <option impratio="100"/>

  <worldbody>
    <body name="object" pos="0.8 0 0.171">
      <geom name="object_box" pos="0 0 0" type="box" size="0.05 0.05 0.05" density="10" rgba="1 0 0 1"/>
      <freejoint name="object_anchor"/>
      <site name="object_site" pos="0 0 0" size="0.011" rgba="0 0 0 0"/>
    </body>

  </worldbody>

</mujoco>
```

https://github.com/user-attachments/assets/9bd36968-45af-4916-a4cc-c4e85aa4fe70

Removing the object:

https://github.com/user-attachments/assets/ba3061bf-46d3-44fe-9c48-3b00e3a9551d

the robot xml:
```
<mujoco model="panda nohand">
  <compiler angle="radian" meshdir="assets" autolimits="true"/>

  <option integrator="implicitfast"/>

  <default>
    <default class="panda">
      <material specular="0.5" shininess="0.25"/>
      <joint armature="0.1" damping="1" axis="0 0 1" range="-2.8973 2.8973"/>
      <general dyntype="none" biastype="affine" ctrlrange="-2.8973 2.8973" forcerange="-87 87"/>
      <default class="finger">
        <joint axis="0 1 0" type="slide" range="0 0.04"/>
      </default>

      <default class="panda_visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2"/>
      </default>
      <default class="panda_collision">
        <geom type="mesh" group="3"/>
      </default>
      <site size="0.001" rgba="0.5 0.5 0.5 0.3" group="4"/>
    </default>

    <!-- dex -->
    <default class="allegro_left">
      <joint axis="0 1 0" damping=".1"/>
      <position kp="1" forcerange="-5 5"/>
      <geom density="800"/>

      <default class="visual">
        <geom type="mesh" contype="0" conaffinity="0" group="2" material="dex_black"/>
        <default class="palm_visual">
          <geom mesh="base_link" material="palm_black"/>
        </default>
        <default class="base_visual">
          <geom mesh="link_0.0" material="base_black"/>
        </default>
        <default class="proximal_visual">
          <geom mesh="link_1.0" material="proximal_black"/>
        </default>
[... 409 lines cut ...]
    <exclude body1="palm" body2="ff_base"/>
    <exclude body1="palm" body2="mf_base"/>
    <exclude body1="palm" body2="rf_base"/>
    <exclude body1="palm" body2="th_base"/>
    <exclude body1="palm" body2="th_proximal"/>
  </contact>

  <!-- <actuator>
    <general class="panda" name="actuator1" joint="joint1" gainprm="4500" biasprm="0 -4500 -450"/>
    <general class="panda" name="actuator2" joint="joint2" gainprm="4500" biasprm="0 -4500 -450"
      ctrlrange="-1.7628 1.7628"/>
    <general class="panda" name="actuator3" joint="joint3" gainprm="3500" biasprm="0 -3500 -350"/>
    <general class="panda" name="actuator4" joint="joint4" gainprm="3500" biasprm="0 -3500 -350"
      ctrlrange="-3.0718 -0.0698"/>
    <general class="panda" name="actuator5" joint="joint5" gainprm="2000" biasprm="0 -2000 -200" forcerange="-12 12"/>
    <general class="panda" name="actuator6" joint="joint6" gainprm="2000" biasprm="0 -2000 -200" forcerange="-12 12"
      ctrlrange="-0.0175 3.7525"/>
    <general class="panda" name="actuator7" joint="joint7" gainprm="2000" biasprm="0 -2000 -200" forcerange="-12 12"/>
  </actuator> -->

  <!-- <keyframe>
    <key name="home" qpos="0 0 0 -1.57079 0 1.57079 -0.7853" ctrl="0 0 0 -1.57079 0 1.57079 -0.7853"/>
  </keyframe> -->
</mujoco>
```

kevinzakka, 2025-03-15:
Hi @gemcollector, I’ll need to check your model and investigate further, but at first glance, I’d recommend creating a minimal XML file that includes only the robot (and the hand, if necessary) for IK computations (aka create a `Configuration` object on the reduced XML). The benefit is faster and more simple solutions which you can copy over to the MjModel/MjData that is simulating robot and objects together.

gemcollector, 2025-03-15:
Hi, kevin! I found that the reason might be I made a mistake about the wrong qpos order. When I fixed this, the IK computation is correct. Thanks for your support! 

But I am still a little bit confused about the `create a Configuration object on the reduced XML`. Could you provide me an example code about the setup ? What is the best way to setup the freejoint object and the robot ? I really appreciate your response!

kevinzakka, 2025-03-15:
@gemcollector Just make a mujoco XML file containing only the robot (no objects) and do mink API calls on this model. Makes sense?

gemcollector, 2025-03-15:
yea, I got it! Thanks for your help!

closed 2025-03-15 by gemcollector