kevinzakka/mink #73: Run IK on a subset of joints

Vittorio-Caggiano, 2025-03-26:
First off, amazing work on this project—I’m a big fan!

I was wondering if there’s already a way to run inverse kinematics (IK) on only a subset of joints. For example, if I have a humanoid robot, I’d like to compute IK just for the right arm while keeping the trunk, legs, and left arm fixed in a specific posture.

Is there an existing way to achieve this? If so, could you provide an example or some guidance on how to set it up?

kevinzakka, 2025-03-26:
Hi @Vittorio-Caggiano, I think there's a few options here:

1. Assign high damping cost on the joints you wish to immobilize. We did this in the [mobile_tidybot](https://github.com/kevinzakka/mink/blob/main/examples/mobile_tidybot.py) example to prevent the base from moving when we want to just move the arm.
2. `solve_ik` will return a velocity vector of size `nv`. Just zero out the velocities for the joints you don't want. Note this is not equivalent to the solution above but kinda works.
3. Create a reduced XML with DOFs only in the parts of interest and run IK on that. You'll have IK running on this reduced model, and have a different MjModel/MjData pair for your actual simulation.

Do these make sense?

Vittorio-Caggiano, 2025-03-26:
Thanks @kevinzakka , those options make a lot of sense!

Will try solutions 1 and 2!

Solution 3 seems the one that might scale better for a large number of DOFs. Do you think that it would be possible to delete DOF with MjSpec at run time?

kevinzakka, 2025-03-26:
I'd probably ask in the MuJoCo github, the MjSpec API is still kinda fluid and I'm not super up-to-date.

Vittorio-Caggiano, 2025-03-26:
sg. Thanks Kevin!

closed 2025-03-26 by Vittorio-Caggiano