kevinzakka/mink #78: error

shaashou, 2025-04-04:
When I run examples/dual_iiwa.py I get this error
Traceback (most recent call last):
  File "/home/ygr/mink/examples/dual_iiwa.py", line 152, in <module>
    vel = mink.solve_ik(
          ^^^^^^^^^^^^^^
  File "/home/ygr/miniconda3/envs/pink/lib/python3.12/site-packages/mink/solve_ik.py", line 101, in solve_ik
    result = qpsolvers.solve_problem(problem, solver=solver, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ygr/miniconda3/envs/pink/lib/python3.12/site-packages/qpsolvers/solve_problem.py", line 81, in solve_problem
    return solve_function[solver](problem, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/ygr/miniconda3/envs/pink/lib/python3.12/site-packages/qpsolvers/solvers/osqp_.py", line 141, in osqp_solve_problem
    "dual_inf_cert": res.dual_inf_cert,
                     ^^^^^^^^^^^^^^^^^
AttributeError: 'types.SimpleNamespace' object has no attribute 'dual_inf_cert'
Thank you very much for helping me solve the error！

ShahhhVihaan, 2025-04-04:
I noticed the same issue with the `humanoid_g1.py` example. The issue is because of API changes in osqp which is used in this and the `dual_iiwa.py` example. Installing `pip install osqp==0.6.3` fixes it.

If you look at the `pyproject.toml` file osqp is an optional dependency of qpsolvers. The qpsolvers repo has also added a version constraint of `<1` https://github.com/qpsolvers/qpsolvers/commit/6f7e566f84dafc2d63d3166f95bb8693dbebcc57

shaashou, 2025-04-06:
> I noticed the same issue with the `humanoid_g1.py` example. The issue is because of API changes in osqp which is used in the `humanoid_g1.py` and the `dual_iiwa.py` example. Installing `pip install osqp==0.6.3` fixes it.我注意到 `humanoid_g1.py` 示例也存在同样的问题。该问题是由于 `humanoid_g1.py` 和 `dual_iiwa.py` 示例中使用的 osqp 中的 API 更改造成的。安装 `pip install osqp==0.6.3` 可以修复此问题。
> 
> If you look at the `pyproject.toml` file osqp is an optional dependency of qpsolvers. The qpsolvers repo has also added a version constraint of `<1` [qpsolvers/qpsolvers@6f7e566](https://github.com/qpsolvers/qpsolvers/commit/6f7e566f84dafc2d63d3166f95bb8693dbebcc57) after their latest release.如果您查看 `pyproject.toml` 文件，osqp 是 qpsolvers 的一个可选依赖项。qpsolvers repo 在其最新版本发布后还添加了 `<1` [qpsolvers/qpsolvers@ 6f7e566 ](https://github.com/qpsolvers/qpsolvers/commit/6f7e566f84dafc2d63d3166f95bb8693dbebcc57) 的版本约束。
 
Thank you, I solved this problem

commit 258116df (2025-04-06): Pin osqp version. Fixes #78.
closed 2025-04-06 by kevinzakka